OpenAMP/libmetal #3: sys_irq_save_disable is broken

asagot, 2017-02-11:
in sys,c, function sys_irq_save_disable() the code shows
`value = mfcpsr() & XIL_EXCEPTION_ALL;`

If interrupts are enabled by default, value will always result in 0x0. 

For example say CPSR = 6000001F (both IRQ and FIR disable bits are low)

0x6000001F & (0x40 | 0x80) wiil yield 0x0000_000

Because int_old_value is initialized to 0x0000_0000 the proceeding If statement will be false and the branch not taken - resulting in the enables remaining ON instead of being turned off

wjliang, 2017-02-14:
Yes, you are right, thanks for picking up this issue.
We can record the value and disabling regardless or change the init value

asagot, 2017-02-14:
ok

closed 2017-02-14 by asagot