OpenAMP/libmetal #4: ZynqMP Demo Doesn't Work On ZCU102

asagot, 2017-02-12:
I ran the libmeal demo that comes prepackaged with PetaLinux 2016.3 and it failed to open an IPI Device

root@Xilinx-ZCU102-2016_3:~#
root@Xilinx-ZCU102-2016_3:~# libmetal-demo
metal: warning:   skipped page size 2097152 - invalid args
metal: error:     device platform:ff340000.ipi not found
metal: error:     metal_irq_register: irq fd -1 is less than 0.
CLIENT> ERROR: Failed to open device ff340000.ipi.
root@Xilinx-ZCU102-2016_3:~#

I had the libmetal demo running on the R5 as well and could see that it registered the devices properly and was waiting for an interrupt for Linux.

What is the proper way to run this demo?

wjliang, 2017-02-12:
hello,

the error message shows cannot find the device. it is because there is no
device node in the DTS. there is liberal DTS example shown in the ug1186
openamp getting started guide . you can add the example DTS device nodes to
your DTS and try again.

best regards,
wendy

asagot, 2017-02-14:
Thanks, that did the trick

closed 2017-02-14 by asagot