OpenAMP/libmetal #6: G++ implementation support

s-rodrigues, 2017-07-06:
Hello,
We were using OpenAMP within the XIlinx tools 2016.2 and compiling it with g++. 
Now we updated the tools versions to 2017.2 and noticed that OpenAMP is not compiling with g++ any more, because of the atomic's present on libmetal.
We have packed it as separate project (wrapper) to compile with gcc and so be able to link it with our g++ compiled project.
Is there going to be a g++ supported implementation? When yes, in which version of Xilinx tools will it be supported?
Thanks

wjliang, 2017-07-06:
Hello,
libmetal is still required to build with c compiler. I think what you mean is to enable your C++ program to use the libmetal/openamp directly.

here is a patch for it:

diff --git a/lib/atomic.h b/lib/atomic.h
index f722d6c..e0cd123 100644
--- a/lib/atomic.h
+++ b/lib/atomic.h
@@ -39,7 +39,11 @@
 #include "metal/config.h"
 
 #if defined(HAVE_STDATOMIC_H) && !defined(__STDC_NO_ATOMICS__)
+#ifdef __cplusplus
+# include <atomic>
+#else
 # include <stdatomic.h>
+#endif
 #elif defined(__GNUC__)
 # include <metal/compiler/gcc/atomic.h>
 #else

Will send out the patch for review

wjliang, 2017-07-06:
I have just sent a patch to the open-amp mailing list to libmetal atomic to support c++
https://groups.google.com/forum/#!topic/open-amp/pXnOMYkOaqY

wjliang, 2018-01-31:
fixes are in the upstream

wjliang, 2018-02-20:
Fixes have been applied

closed 2018-02-20 by wjliang