OpenAMP/libmetal #8: ZynqMP Demo Doesn't Work On ZCU102

kf6uzf, 2017-12-07:
I am having trouble running the libmetal demo on Linux. It seems the IPI device is not defined.
During boot the drivers say the IPI device is not defined.

Here is the boot message related to the remoteproc initialization:
[    5.099446] zynqmp_r5_remoteproc ff9a0100.zynqmp_r5_rproc: RPU core_conf: split0
[    5.106917] zynqmp_r5_remoteproc ff9a0100.zynqmp_r5_rproc: **IPI resource is not specified**.
[    5.116141] remoteproc remoteproc0: ff9a0100.zynqmp_r5_rproc is available
[    5.122966] remoteproc remoteproc0: Note: remoteproc is still under development and considered experimental.
[    5.132820] remoteproc remoteproc0: THE BINARY FORMAT IS NOT YET FINALIZED, and backward compatibility isn't yet guaranteed.
[    5.144174] zynqmp_r5_remoteproc ff9a0200.zynqmp_r5_rproc: RPU core_conf: split1
[    5.152017] remoteproc remoteproc1: ff9a0200.zynqmp_r5_rproc is available
[    5.158828] remoteproc remoteproc1: Note: remoteproc is still under development and considered experimental.
[    5.168691] remoteproc remoteproc1: THE BINARY FORMAT IS NOT YET FINALIZED, and backward compatibility isn't yet guaranteed.

When loading the R5 code (linmetal-AMP-demo) it emits this:
[   20.922576] remoteproc remoteproc0: powering up ff9a0100.zynqmp_r5_rproc
[   20.931915] remoteproc remoteproc0: Booting fw image libmetal-demo, size 541332
[   20.941412] zynqmp_r5_remoteproc ff9a0100.zynqmp_r5_rproc: RPU boot from TCM.
[  registering: 0, **name=ff310000.ipi**
   registering: 1, name=3ed80000.shm
   registering: 2, name=ff110000.ttc

SERVER> ====== libmetal demo: shared memory ======

SERVER> Wait for shared memory demo to start.
 20.951156] remoteproc remoteproc0: remote processor ff9a0100.zynqmp_r5_rproc is now up

The libmetal demo common.h file has this:

/* Devices names */
#define BUS_NAME        "generic"
#define IPI_DEV_NAME    **"ff310000.ipi"**
#define SHM_DEV_NAME    "3ed80000.shm"
#define TTC_DEV_NAME    "ff110000.ttc"

My linux demo application common.h files has this:
#define BUS_NAME        "platform"
#define IPI_DEV_NAME    **"ff340000.ipi"**
#define SHM_DEV_NAME    "3ed80000.shm"
#define TTC_DEV_NAME    "ff110000.timer"

It seem the R5 app is using a different address for IPI than the Linux app. My device file (attached) also uses the **"ff340000.ipi"** definition.

When running the Linux app I get this"
CLIENT> ****** libmetal demo: shared memory ******
metal: info:      metal_uio_dev_open: No IRQ for device 3ed80000.shm.
CLIENT> Setting up shared memory demo.
CLIENT> Starting shared memory demo.
CLIENT> Sending message: Hello World - libmetal shared memory demo
CLIENT> Message Received: Hello World - libmetal shared memory demo
CLIENT> Shared memory demo: Passed.
CLIENT> ****** libmetal demo: atomic operation over shared memory ******
metal: info:      metal_uio_dev_open: No IRQ for device 3ed80000.shm.
metal: error:     failed to bind ff340000.ipi to uio_pdrv_genirq
metal: error:     metal_irq_register: irq fd -1 is less than 0.
CLIENT> ERROR: Failed to open device ff340000.ipi.
CLIENT> ERROR: shared memory atomic demo failed.

Any help would be appreciated

Attached is my device file
[system-user.dtsi.txt](https://github.com/OpenAMP/libmetal/files/1540627/system-user.dtsi.txt)

wjliang, 2017-12-08:
from your .dtsi file, you have specified interrupts in the test_r50 and test_r51, the ipi uio also uses the same interrupt which is the IPI interrupt. The UIO doesn't support interrupt sharing.
you can remove "interrupt-parent = <&gic>; interrupts = <0 29 4>;" from the test_r50 and test_r51

kf6uzf, 2017-12-08:
Removed the interrupt lines and still the same problem. Why does the IPI_DEV_NAME value differ in the rpu application (ff310000.ipi) as opposed to the Linux app (ff340000.ipi). The other values are the same (SHM_DEV_NAME & TTC_DEV_NAME).
Also, why do I get the " IPI resource is not specified." message during boot?

wjliang, 2017-12-08:
The Linux is using registers of PL0 IPI agent and the RPU0 is using registers of RPU0 IPI agent. you can check the TRM on the IPI description for more details
in the remoteproc boot code, it will check if you have IPI specified, if yes, it will setup IPI notification. If you use RPMsg in kernel space, you will need to have IPI resource in the RPU remoteproc node. But if your userspace directly use IPI and the remoteproc is just to load firmware, you don't set the IPI resource.
Please check your /sys/bus/platform, see if you see IPI devices there, the IPI device name needs to match the one there

kf6uzf, 2018-01-03:
First of all I would like to thank you from all your help so far...but
I am still unable to run the Libmetal AMP Demo (following the instructions in UG1186 (v2017.3) October 18, 2017).
I am still getting the metal: error: failed to bind ff340000.ipi to uio_pdrv_genirq error.
I tried what you suggested (removing the entries in your previous post) but then got IPI errors on boot.
There is a an device in /sys/bus/platform/devices called ff340000.ipi.
When I query the defined interupts for that device I get this
`root@zcu102:/sys/bus/platform/devices/ff340000.ipi# cat of_node/interrupts | od -x
0000000 0000 0000 0000 1d00 0000 0400
0000014
`
which matches the deploy/images/plnx_aarch64/plnx_aarch64-system.dts entries for the ipi device. Example
`ipi@ff340000 {
			compatible = "ipi_uio";
			reg = <0x0 0xff340000 0x0 0x1000>;
			interrupt-parent = <0x4>;
			interrupts = <0x0 0x1d 0x4>;
		};`
My system-user.dtsi has this
`ipi: ipi@ff340000  {
             compatible = "ipi_uio";
             reg = <0x0 0xff340000 0x0 0x1000>;
             interrupt-parent = <&gic>;
             interrupts = <0 29 4>;
         };`
Why are they different?
I have followed in the instruction in the UG1186 to a tee and have done it many times to exclude any errors.
I there anyone else that is having the same problem?
Thanks again and I apologize if I am being annoying.
Let me know if you need further details.
Brad

wjliang, 2018-01-03:
Hi Brad,
Could you paste the DTS generated by your DTB used in your Linux bringup?
I am not seeing this issue?

you can also try manually bind the ipi device with UIO driver by directly accessing the sysfs:
# echo "uio_pdrv_genirq" > /sys/bus/platform/devices/ff340000.ipi/driver_override
# echo "ff340000.ipi" > /sys/bus/platform/drivers/uio_pdrv_genirq/bind
You should see the same issue without using libmetal application.

I still think it related to DTS at this point

kf6uzf, 2018-01-03:
Did you mean provide the DTB from the DTS and DTSI files. Either way I include the entire folder.
Also included is my system-user.dts for reference.

[system-user.dtsi.zip](https://github.com/OpenAMP/libmetal/files/1601590/system-user.dtsi.zip)

[device-tree-generation-folder.zip](https://github.com/OpenAMP/libmetal/files/1601584/device-tree-generation-folder.zip)

Thanks again!

kf6uzf, 2018-01-03:
I ran the second command line command above (bind) and the is what was output

`root@zcu102:~# echo "ff340000.ipi" > /sys/bus/platform/drivers/uio_pdrv_genirq/bind
[   58.992558] genirq: Flags mismatch irq 55. 00000004 (ipi) vs. 00000084 (ff9a0100.zynqmp_r5_rproc)
[   59.003773] uio_pdrv_genirq ff340000.ipi: unable to register uio device
[   59.012520] uio_pdrv_genirq: probe of ff340000.ipi failed with error -16
-sh: echo: write error: No such device
`

wjliang, 2018-01-03:
from your system-user.dtsi,
you have used the same interrupt as the ipi device node, UIO doesn't support shared interrupt.
You will need to remove the IPI and the interrupts in the remoteproc device node:
"
in the remoteproc boot code, it will check if you have IPI specified, if yes, it will setup IPI notification. If you use RPMsg in kernel space, you will need to have IPI resource in the RPU remoteproc node. But if your userspace directly use IPI and the remoteproc is just to load firmware, you don't set the IPI resource.
Please check your /sys/bus/platform, see if you see IPI devices there, the IPI device name needs to match the one there

Also in your DTS,
"
reg = <0x0 0xff9a0100 0 0x100>, <0x0 0xff340000 0 0x100>, <0x0 0xff9a0000 0 0x100>;
reg-names = "rpu_base", "rpu_glbl_base";
"

you will also need to remove ", <0x0 0xff340000 0 0x100>", that's the IPI PL0 agent address, you use it as "rpu_glbl_base".

You will need to remove the IPI from reg and reg-names for both remoteproc nodes.
"

kf6uzf, 2018-01-03:
OK, I have made the changes you requested and *most* of the test pass except one.
Thank you for getting me this far. I don't know hwy when you had mentioned the steps above that my process didn't work. I will pay more attention in the future.

The output of the tests are below with the "shared mem throughput demo" test failing. Amy ideas on why only this test failed based on the files I have sent you.

Thanks

CLIENT>` ****** libmetal demo: shared memory ******
metal: info:      metal_uio_dev_open: No IRQ for device 3ed80000.shm.
CLIENT> Setting up shared memory demo.
CLIENT> Starting shared memory demo.
CLIENT> Sending message: Hello World - libmetal shared memory demo
CLIENT> Message Received: Hello World - libmetal shared memory demo
CLIENT> Shared memory demo: Passed.
CLIENT> ****** libmetal demo: atomic operation over shared memory ******
metal: info:      metal_uio_dev_open: No IRQ for device 3ed80000.shm.
CLIENT> Starting atomic shared memory task.
CLIENT> shm atomic demo PASSED!
CLIENT> ****** libmetal demo: IPI and shared memory ******
metal: info:      metal_uio_dev_open: No IRQ for device 3ed80000.shm.
CLIENT> Start echo flood testing....
CLIENT> Sending msgs to the remote.
CLIENT> Waiting for messages to echo back and verify.
CLIENT> Kick remote to notify shutdown message sent...
CLIENT> Total packages: 1024, time_avg = 0s, 1545ns
CLIENT> ****** libmetal demo: IPI latency ******
metal: info:      metal_uio_dev_open: No IRQ for device 3ed80000.shm.
CLIENT> Starting IPI latency task
CLIENT> IPI latency result with 1000 iterations:
CLIENT> APU to RPU average latency: 34 ns 
CLIENT> RPU to APU average latency: 34 ns 
CLIENT> Finished IPI latency task
CLIENT> ****** libmetal demo: shared memory latency ******
metal: info:      metal_uio_dev_open: No IRQ for device 3ed80000.shm.
CLIENT> Starting IPI latency task
CLIENT> package size 16 latency result:
CLIENT> APU to RPU average latency: 37 ns 
CLIENT> RPU to APU average latency: 0 ns 
CLIENT> package size 32 latency result:
CLIENT> APU to RPU average latency: 6 ns 
CLIENT> RPU to APU average latency: 14 ns 
CLIENT> package size 64 latency result:
CLIENT> APU to RPU average latency: 17 ns 
CLIENT> RPU to APU average latency: 23 ns 
CLIENT> package size 128 latency result:
CLIENT> APU to RPU average latency: 13 ns 
CLIENT> RPU to APU average latency: 33 ns 
CLIENT> package size 256 latency result:
CLIENT> APU to RPU average latency: 17 ns 
CLIENT> RPU to APU average latency: 10 ns 
CLIENT> package size 512 latency result:
CLIENT> APU to RPU average latency: 32 ns 
CLIENT> RPU to APU average latency: 34 ns 
CLIENT> package size 1024 latency result:
CLIENT> APU to RPU average latency: 40 ns 
CLIENT> RPU to APU average latency: 21 ns 
CLIENT> Finished shared memory latency task
CLIENT> ****** libmetal demo: shared memory throughput ******
metal: info:      metal_uio_dev_open: No IRQ for device 3ed80000.shm.
CLIENT> Starting shared mem throughput demo
CLIENT> ERROR: [0]failed to get rx offset: 0x3f780000, 0x3ef80004.
CLIENT> ERROR: shared memory throughput demo failed.

wjliang, 2018-01-04:
The libmetal shared memory test uses quite some memory, and thus needs to reserve more memory.
I have checked the ug1186, the device tree memory example is wrong in the DTS, we will fix. And thanks for pointing it out.

Here is the change:
reserved-memory {
                #address-cells = <2>;
                #size-cells = <2>;
                ranges;
                rproc_0_reserved: rproc@3ed000000 {
                        no-map;
                        reg = <0x0 0x3ed00000 0x0 0x2000000>;
                };
        };
 shm0: shm@0 {
                        compatible = "shm";
                        reg = <0x0 0x3ed80000 0x0 0x1000000>;
                };

kf6uzf, 2018-01-04:
The last DTS change has solve my issue and the Libmetal AMP Demo test application runs to completion.
I did notice that I cannot run the test application more than once without a reboot but I can figure that out on my own and will forward my findings to you.
Thank you again for your help and your **patience**. I really appreciate both!.

closed 2018-01-04 by kf6uzf

wjliang, 2018-01-04:
did you need to reboot the whole ZCU102 board or just the RPU? 
Once the RPU application finishes running , it will not automatically started, you can use remoteproc sysfs to restart the RPU only.

But if you need to restart the whole board that will be a different issue.

kf6uzf, 2018-01-10:
I have confirmed that only an RPU restart is required to re-run the Linux demo app which makes sense because the RPU firmware's main function exits after the tests are complete.
Maybe a note in the UG1186 explaining this behavior would benefit others when running the tests.
Just a suggestion....

Thanks again!

justdoGIT, 2019-01-22:
Hi, I am trying to implement the **ZynqMP Linux Master running on APU with RPMsg in userspace and 2 RPU slaves**.
the dts file compiled from dtb is attached here.
I've followed the steps in ug1186 but I am not able to boot r5 core.
the bootgen.bif file is also attached.
[bootgen.txt](https://github.com/OpenAMP/libmetal/files/2783116/bootgen.txt)
[device-tree.txt](https://github.com/OpenAMP/libmetal/files/2783131/device-tree.txt)
I didn't see any logs related to r5 while booting
any pointers will be helpful

likithiam202, 2020-12-16:
@kf6uzf 
Could you please share the final system-user.dtsi file which worked for you ?

Thanks