OpenAMP/libmetal #13: libmetal APIs for share memory transactions

arnop2, 2018-02-20:
Hello,

This issue is a copy past of discussion around memory management in OpenAMP and libmetal.
Aim of this issue is to share a proposal to offer an unique libmetal API for the memory management 

Nothing very complex, just a proposal to hand over memory management in libmetal  porting layer.

Memory management in term of feature:
-	variable/structure used in openamp:
    o	Dynamically  allocated
    o	Statically allocated
=> not treated here but should could treated using a compilation switch.
-	Shared memory to map. 
    o	Allocate and/or  register a memory region
    o	map/unmap of a memory region
    o	read/write/set access to the memory

Shared memory access type:
-	basic memory mapping (no MMU /IOMMU, no cache…) 
	va = pa
-	MMU/IOMMU with simple a memory translation
	Need to manage translation between va and pa
-	MMU/IOMMU, specific buses with specific memory management ( paging, cache…)
	Specific memory  management.

Overview of the shared Memory management used by OpenAMP, provided by libmetal
-	io region  :  memory and i/o register access 
	provide ops to access to memory
libmetal/include/io.h
-	dev_mem : memory device access through bus abstraction 
libmetal/include/device.h
=>based on io_region
-	Shmem:  Registration and open of shared memory region
libmetal/include/shmem.h
=> based on  io region.
-	cache management: used in OpenAMP for flush and invalidate cache
               => libmetal/include/cache.h

Memory management in rpmsg, virtio and rproc frameworks implemented in OpenAMP  
	Mainly the io_region

Concerns about current implementation:
-	Complex memory management is exposed at OpenAMP level for all platforms.
-	OpenAMP should be memory management agnostic. 
-	Libmetal as porting layer should implement associated complexity level.

Proposal:
-	Define an unique generic libmetal API for shared memory management. 
-	Clients(OpenAMP and application) uses this API to access to shared memory.
-	Shared memory regions will be registered.  Application is in charge of allocating or mapping the memory associated.
-	Shared memory  is identified by an handle.

API proposed ( first draft under discussion)
	Based on principles described above.
-	An unique libmetal API for shared memory management : shmem.h 
=> Current devmem /io region and cache APIs for shared memory management no more exposed, but implemented in libmetal depending on platform.
-	Proposed API is inspirited from  io_region API.
-	OpenAMP & application access to shared memory using a memory region handle and an offset. 
-	Every action on shared memory is done by  libmetal.

/** Generic shared memory data structure that could be optimized depending on platform and OS. */
extern struct metal_generic_mem;

/* Example of  a declaration to remap it on io memory region*/
struct metal_generic_shmem {
               const char                         *name;
               metal_phys_addr_t        pa;
               metal_phys_addr_t        size;
               void * metal_io_region ; 
               struct metal_list  node;
};

/* memory region Registration( carveouts, vrings, buffers...) */
extern int metal_shmem_register_generic(struct metal_generic_shmem *shmem);

/* Retrieve registered memory region by name or address */
extern struct metal_generic_mem *metal_shmem_open_by_name(const char *name, size_t size,
                                                struct metal_io_region **io);
extern struct metal_generic_mem *metal_shmem_open_by_addr(metal_phys_addr_t pa, size_t size,
                                                struct metal_io_region **io);
extern void metal_shmem_block_close(struct metal_generic_shmem *shmem);

/* Get physical address */
extern metal_phys_addr_t pa  metal_shmem_to_phy (struct metal_generic_shmem * shmem,  unsigned long offset);

/* Memory region access */
extern uint64_t metal_shmem_read(struct metal_generic_mem *mem, unsigned long offset, int width);
extern int metal_shmem_write(struct metal_generic_shmem *shmem, unsigned long offset,  uint64_t value, int width);
extern int metal_shmem_block_read(struct metal_generic_shmem *shmem, unsigned long offset, void *restrict dst, int len);
extern int metal_shmem_block_write(struct metal_generic_shmem *shmem, unsigned long offset, const void *restrict src, int len);
extern int metal_shmem_block_set(struct metal_generic_shmem *shmem, unsigned long offset, unsigned char value, int len);

BR,
Arnaud

wjliang, 2018-02-21:
* the change to "struct metal_generic_shmem" is adding the pa and the size, however, do we need to have the "pa" and "size"? as the io region already have those two attributes
* metal_shmem_open_by_addr(), I think it good to have searching shared memory from address API
* shared memory access operations _write/_read, _block_write/_block_read, they still uses "offset", how about use "va" pointer instead?
* from the definition of "struct metal_generic_shmem", it looks like io is still necessary of the generic_shmem. I am still thinking about if we should add new shmem operation or extend the existing _io_region functions. Or should the "struct meta_generic_shmem" to have "va" as attribute as the not every shared memory have I/O region, or do you suggests I/O is always required?

arnop2, 2018-02-22:
>- The change to "struct metal_generic_shmem" is adding the pa and the size, however, do we need to >have the "pa" and "size"? as the io region already have those two attributes
Right, in case io region is used no need, if no io region used, will need it. So to be adapted depending on platform.
>- metal_shmem_open_by_addr(), I think it good to have searching shared memory from address API
You mean in addition of this API, metal_shmem_open_by_add?
>- shared memory access operations _write/_read, _block_write/_block_read, they still uses "offset", how >about use "va" pointer instead?
ok for this, this could simplify for platform that have a simple translation between va an pa ( or va= pa).  on the other side that means a va_to_offset for io_remap.
Please notice that metal_io_ops use offset, not a va  (e.g. metal_io_write).
> - from the definition of "struct metal_generic_shmem", it looks like io is still necessary of the >generic_shmem. I am still thinking about if we should add new shmem operation or extend the [...]
struct metal_generic_shmem is described in my comment is an example. Idea is to be able to define
it depending on the platform need. As example for a processor without logical address:
 struct metal_generic_shmem {
const char *name;
metal_phys_addr_t pa;
metal_phys_addr_t size;
struct metal_list node;
};
platform with logical address management va:
 struct metal_generic_shmem {
const char *name;
void va* ;
metal_phys_addr_t pa;
metal_phys_addr_t size;
struct metal_list node;
};
platform with with I/O region management:
 struct metal_generic_shmem {
const char *name;
void * metal_io_region ;
struct metal_list node;
};
As discussed in sync-up, i will have a look to extend io_region API instead of shmem

wjliang, 2018-02-22:
Hi Arnaud,

I have got one comments from Cyril, which is a valid point, that is the I/O region also provides ordering, that is when we use I/O region to do memory access (_write/_read/_block_write/_block_read), the I/O region operation do the ordering for us even for normal memory. We may need to rethink about how we use I/O in the RPMsg, virtio implementation.
At the moment, the I/O feature is not fully used in the virtio implementation, we manually put memory barrier there, i should think about to use metal_io_read/write, e.g. in virtqueue_notification, when updated used vring index, we should use metal_io_write.

How do you think?

Thanks,
Wendy

arnopo, 2023-05-09:
I close this one. memory management optimisation should be part of a new Issue. this one is not up to date

closed 2023-05-09 by arnopo