OpenAMP/libmetal #14: metal_io_phys(): bug on arm 64-bit when io->page_shift is equal to 64

rfeneon, 2018-03-02:
Hi, I have encountered an issue with the metal_io_phys() function on arm 64-bit target when io->page_shift is equal to 64, the result of the following shift operation lets 'offset' unchanged, so 'page' equals 'offset' instead of 0:
unsigned long page = offset >> io->page_shift;

It seems that according to the C standard the behaviour of the right shift operator is undefined when the shift operand is greater or equal to the length in bits of the value to be shifted, so it's not a bug in gcc.

I have corrected the problem with the following code:

static inline metal_phys_addr_t
metal_io_phys(struct metal_io_region *io, unsigned long offset)
{
  unsigned long page;
  if (io->page_shift >= 8*sizeof(unsigned long))
    page = 0;
  else
    page = offset >> io->page_shift;
...

Thanks

wjliang, 2018-03-06:
Thanks for pointing out the issue, will you have a patch?

How about the following change?

static inline metal_phys_addr_t
metal_io_phys(struct metal_io_region *io, unsigned long offset)
{
unsigned long page = (io->page_shift >= 8 * sizeof(offset) ? 0 : offset >> io->page_shift ;

Thanks,
Wendy

rfeneon, 2018-03-06:
Your change looks ok to me.
I don't have a patch, will you make the change or do you prefer that I sent one?
thanks

wjliang, 2018-03-13:
I have created a pull request: https://github.com/OpenAMP/libmetal/pull/15
Please comment.
Thanks,
Wendy

closed 2018-03-21 by wjliang