OpenAMP/libmetal #38: Have logging on zephyr use sys_log mechanism

galak, 2018-04-06:
We currently use printk, but should tie into the proper logging system in zephyr via include/logging/sys_log.h.

skordal, 2018-04-10:
This won't directly work with the current libmetal logging API. The problem is that the Zephyr syslog APIs are macros (`SYS_LOG_ERR()`/`SYS_LOG_WRN`/etc), so we can't call them correctly from the libmetal log handler.

However, we could call the `SYS_LOG_BACKEND_FN` instead of `printk` from the log handler. We won't get color support or the consistent logging format provided by the `SYS_LOG_` macros, but we can support custom Zephyr logging functions (`syslog_hook`).

galak, 2018-04-10:
I'll add a comment to https://github.com/zephyrproject-rtos/zephyr/issues/6186 to improve the logging API in zephyr to allow us to connect this more directly.

galak, 2018-04-10:
@skordal can we use or tie into LOG_NO_COLOR()?

skordal, 2018-04-11:
That won't work either, we will need a function that can take a `va_list` argument, since we need to forward the arguments passed to the libmetal log handler function.

(My suggestion about using the SYS_LOG_BACKEND_FN actually won't work either, for the same reason).

closed 2018-09-24 by wjliang