OpenAMP/libmetal #65: GNU extension: void pointer arithmetic

mped-oticon, 2018-09-25:
libmetal is currently relying on a GNU C extension, namely void-ptr arithmetic, which is a violation across all ISO C standards.
Worse, this makes libmetal non-edible by Clang.

```
libmetal/lib/io.c:54:7: error: wrong type argument to increment [-Werror=pointer-arith]
libmetal/lib/io.c:58:39: error: pointer of type ‘void *’ used in arithmetic [-Werror=pointer-arith]
libmetal/lib/io.c:62:23: error: wrong type argument to increment [-Werror=pointer-arith]
libmetal/lib/io.c:90:7: error: wrong type argument to increment [-Werror=pointer-arith]
libmetal/lib/io.c:94:10: error: pointer of type ‘void *’ used in arithmetic [-Werror=pointer-arith]
libmetal/lib/io.c:97:30: error: wrong type argument to increment [-Werror=pointer-arith]
```

kind regards,
Mark

wjliang, 2018-09-25:
Thanks to bring issues up, it will be good to have libmetal work with clang.
For the compilation error reported, it doesn't look like it matches the HEAD of master.
Could you try the master branch to see if it is still an issue for you?

Best Regards,
Wendy

mped-oticon, 2018-09-26:
Hi Wendy, thanks for your response!

Shame on me - I should have taken a better look before.
It appears fixed already: https://github.com/OpenAMP/libmetal/commit/efc33fe79d701245d26e0500abd44d2ed49b8045 

Context: Observed from Zephyr's copy of libmetal. (I'll try to get Zephyr updated to latest libmetal).

Regards
Mark

closed 2018-09-26 by mped-oticon