OpenAMP/libmetal #66: compile libmetal for zynq linux,but failed

oska874, 2018-10-11:
I tried to compile libmetal for zynq linux ,but it fails, show that:

```
➜  build-libmetal git:(master) ✗ cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake/platforms/zynq7-linux.cmake
-- Build type:  Debug
-- Host:    Linux/x86_64
-- Target:  Linux/arm
-- Machine: Generic
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE)
-- Could NOT find LIBSYSFS (missing:  LIBSYSFS_LIBRARY)
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find LibRt (missing: LIBRT_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindLibRt.cmake:36 (find_package_handle_standard_args)
  cmake/depends.cmake:24 (find_package)
  CMakeLists.txt:18 (include)

-- Configuring incomplete, errors occurred!
See also "/home/osboxes/src/tool/libmetal/build-libmetal/CMakeFiles/CMakeOutput.log".
See also "/home/osboxes/src/tool/libmetal/build-libmetal/CMakeFiles/CMakeError.log".
```

I use `arm-linux-gnueabihf-gcc` as cross-compiler,and I have modified it in `zynq7-linux.cmake`.
How to solve this problem?

oska874, 2018-10-11:
I compile it for x86 is ok.

wjliang, 2018-10-11:
The error suggested there is no libRt in your rootfs, we use yocto to build the library and the yocto helps to solve the dependencies, if you are not using yocto, you will need to build libRt into your cross compiled sysroot, and then you will need to use that sysroot to compile libmetal.

oska874, 2018-10-12:
The guide for libmetal and openamp is a little poor somewhat.

wjliang, 2018-10-12:
> How to build libmetal with yocto? I downloaded meta-openamp and yocto, but do not know how to config yocto?

you can use meta-openamp to build libmetal, the package is libmetal

wjliang, 2018-10-12:
> The guide to how to use libmetal and openamp is a little poor somewhat. Makes confusion.

The guide is more for the baremetal and Linux host. Agree, should be improved to cover cross Linux compilation

oska874, 2018-10-12:
> > How to build libmetal with yocto? I downloaded meta-openamp and yocto, but do not know how to config yocto?
> 
> you can use meta-openamp to build libmetal, the package is libmetal

Add meta-openamp layer to bblayers.conf and execute `bitbake libmetal` ?
Should I modify `MACHINE` in `conf/local.conf` ?

oska874, 2018-10-12:
I did not use yocto directly to build zynq linux, so how to **just** build libmetal and openamp library?

wjliang, 2018-10-12:
> Add meta-openamp layer to bblayers.conf and execute `bitbake libmetal` ?
> Should I modify `MACHINE` in `conf/local.conf` ?(`zynq` is not defined)

set `MACHINE` to `zc702-zynq7`, you can build core-image-minimal, it will generate the rootfs for you.

wjliang, 2018-10-12:
> I did not use yocto directly to build zynq linux, so how to **just** build libmetal and openamp library?
> After openamp library generated, I will program with the library.

You will need to build the the dependents, for libmetal, you will need to build librt and libsysfs with your build tool before building libmetal

waqarrashid33, 2019-08-29:
You are probably missing the following dependencies..
`sudo apt-get install doxygen`
`sudo apt-get install libsysfs-dev`

MrLi92, 2020-03-24:
> You are probably missing the following dependencies..
> `sudo apt-get install doxygen`
> `sudo apt-get install libsysfs-dev`

it's work ，thinks。

closed 2020-04-10 by arnopo

MiguelPond, 2024-05-26:
On Fedora 40 that dependency libsysfs-devel fixed this issue for me