OpenAMP/libmetal #69: FreeRTOS at higher tick rates bug

DaveTrussel, 2018-11-29:
In `libmetal/lib/system/freertos/sleep.h`
```
static inline int __metal_sleep_usec(unsigned int usec)
{
	const TickType_t xDelay = usec / portTICK_PERIOD_MS;
	vTaskDelay(xDelay);
	return 0;
}
```

If the `configTICK_RATE_HZ` of is bigger than 1000 the `portTICK_PERIOD_MS` is 0 and you get division by zero (not recommended).
Better use `seconds * configTICK_RATE_HZ` to calculate how many ticks to sleep.

wjliang, 2018-12-03:
> If the `configTICK_RATE_HZ` of is bigger than 1000 the `portTICK_PERIOD_MS` is 0 and you get division by zero (not recommended).

I just check the source code of Xilinx FreeRTOS porting, confirmed what you pointed out. Agree, should change to what you suggested.

Best Regards,
Wendy

edmooring, 2020-12-17:
This is indeed an issue. We should probably be using pdMS_TO_TICKS() instead.

edmooring, 2020-12-18:
See #143 for a possible fix.

closed 2021-04-28 by arnopo