OpenAMP/libmetal #74: Atomic Flag - Use of deleted Function - Petalinux 2018.3

JimmyTask, 2019-01-14:
Hello, I'm using petalinux 2018.3 and trying to build a custom application that uses the Zynq RF Data Converter which relies on the libmetal library.  When I build my application with my RFADC.h header file above a #include future then I get the following error:

`In file included from /opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/metal/atomic.h:21:0,
                 from /opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/metal/io.h:21,
                 from /opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/metal/device.h:16,
                 from /opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/xrfdc.h:214,
                 from include/RFADC.h:27,
                 from include/ZCU111DevScanner.h:32,
                 from src/ZCU111DevScanner.cpp:12:
/opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/c++/7.3.0/future:320:43: error: use of deleted function ‘std::atomic_flag::atomic_flag(const std::atomic_flag&)’
       atomic_flag          _M_retrieved = ATOMIC_FLAG_INIT;

In file included from /opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/c++/7.3.0/bits/shared_ptr_atomic.h:33:0,
                 from /opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/c++/7.3.0/memory:82,
                 from /opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/c++/7.3.0/thread:39,
                 from ../blackmore_common/include/CommonMisc.h:24,
                 from ../blackmore_common/include/BlackmoreMessage.h:16,
                 from include/ZCU111DevScanner.h:13,
                 from src/ZCU111DevScanner.cpp:12:
/opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/c++/7.3.0/bits/atomic_base.h:164:5: note: declared here
     atomic_flag(const atomic_flag&) = delete;
     ^~~~~~~~~~~
/opt/petalinux/2018.3/sysroots/aarch64-xilinx-linux/usr/include/c++/7.3.0/bits/atomic_base.h:169:15: note:   after user-defined conversion: constexpr std::atomic_flag::atomic_flag(bool)
     constexpr atomic_flag(bool __i) noexcept
               ^~~~~~~~~~~

`

But if I move the #include RFADC.h to be below the #include future then it builds successfully.  I understand that this works, but it seems like a hack and I cannot count on someone to put the header files in the proper order in the future.

Is there any way to solve this so that the application will build successfully independently on where the header files are (with regards to libmetal)?  Thanks in advance.

wjliang, 2019-01-17:
Just try to understand this issue, we have compiled the library with c++ for baremetal, but I haven't tried with Linux c++, could you confirm that this issue happens when you compile a Linux c++ application?
Will try this and get back to you.

Best Regards,
Wendy

JimmyTask, 2019-01-18:
That's correct, I am using this in a Linux application (Petalinux 2018.3).

wjliang, 2019-02-04:
Patches has been applied to libmetal repo, closing this issue, if the issue persist, please reopen it.

closed 2019-02-04 by wjliang