OpenAMP/libmetal #97: C++ Support in Petalinux 2019.1

hgiannopoulos, 2019-06-27:
I'm having problems getting a super basic C++ applet with Libmetal to compile in Petalinux 2019.1 for the RFSoC. 

My Super Simple Example app is:
```
#include <iostream>              
#include <metal/io.h> 
                                                                                                                                                                                                                                           
int main(int argc, char *argv[]){                                                                                                    
                                                                                                                                     
   // struct metal_device * device;  
   int i;    

   std::cout << "Hello World!" << std::endl;     

                                                                           
   return 0;                                                                                                                       
}
```

This build fine in 2018.3 but now in 2019.1 It throws a number of errors relating to atomic:
```
/usr/include/metal/io.h:42:5: error: 'memory_order' has not been declared
     memory_order order,
     ^~~~~~~~~~~~
/usr/include/metal/io.h:47:6: error: 'memory_order' has not been declared
      memory_order order,
      ^~~~~~~~~~~~
/usr/include/metal/io.h:52:5: error: 'memory_order' has not been declared
     memory_order order,
     ^~~~~~~~~~~~
/usr/include/metal/io.h:57:6: error: 'memory_order' has not been declared
      memory_order order,
      ^~~~~~~~~~~~
/usr/include/metal/io.h:62:6: error: 'memory_order' has not been declared
      memory_order order,
      ^~~~~~~~~~~~
/usr/include/metal/io.h:235:8: error: 'memory_order' has not been declared
        memory_order order, int width)
        ^~~~~~~~~~~~
/usr/include/metal/io.h: In function 'uint64_t metal_io_read(metal_io_region*, long unsigned int, int, int)':
/usr/include/metal/io.h:241:25: error: 'atomic_uchar' was not declared in this scope
  else if (ptr && sizeof(atomic_uchar) == width)
                         ^~~~~~~~~~~~
/usr/include/metal/io.h:241:25: note: suggested alternative:
In file included from /usr/include/metal/atomic.h:21,
                 from test.cpp:2:
/usr/include/c++/8.2.0/atomic:877:34: note:   'std::atomic_uchar'
   typedef atomic<unsigned char>  atomic_uchar;
                                  ^~~~~~~~~~~~
In file included from test.cpp:3:
/usr/include/metal/io.h:242:46: error: expected primary-expression before ')' token
   return atomic_load_explicit((atomic_uchar *)ptr, order);
                                              ^
/usr/include/metal/io.h:242:10: error: 'atomic_load_explicit' was not declared in this scope
   return atomic_load_explicit((atomic_uchar *)ptr, order);
          ^~~~~~~~~~~~~~~~~~~~
/usr/include/metal/io.h:242:10: note: suggested alternative:
In file included from /usr/include/metal/atomic.h:21,
                 from test.cpp:2:
/usr/include/c++/8.2.0/atomic:1089:5: note:   'std::atomic_load_explicit'
     atomic_load_explicit(const volatile atomic<_ITp>* __a,
...
```

 Is anyone else able to replicate the behavior I'm seeing?

edmooring, 2019-07-03:
I can replicate it, even with the PetaLinux 2018.3 tool chain. I believe the fix is to revert 81f85dc244d05448e26aa674301065a1220b4539. This will prevent mixing the C and C++ definitions for atomic types.

wjliang, 2019-07-10:
the reversion not to use <atomic> but use the atomic operations defined in libmetal can be a workaround.
It will be better to fix it properly to be  able to use the atomic defined by g++. instead of use 
```
#ifdef __cplusplus
extern "C" {
#endif
...
#ifdef __cplusplus
}
#endif
```
to cover the whole file,  it is better to use it for c specific part and for the C++, it is better to use
```
# if defined(__cplusplus)
#endif
```
to cover it.

edmooring, 2019-07-10:
That still won't work, because memory_order is in a different namespace, so any C functions that use the type will not be able to find it. The standards committees for C and C++ sort of dropped the ball on this one. See http://www.open-std.org/JTC1/SC22/WG21/docs/papers/2018/p0943r0.html for more on this topic and a possible solution. 

For now, I favor reverting the commit to let people keep compiling libmetal applications in C++.

closed 2019-11-18 by edmooring