OpenAMP/libmetal #99: Compiling for baremetal, do we need to change the cmake files for different platforms?

waqarrashid33, 2019-08-29:
Hello everyone,
I am just starting with libmetal and openamp so pardon me if this question is very basic.

So I need to compiled some of the examples in openamp, especially echo one. In order to run it on RPC I have to compiled using the baremetal compilation method. I am working on RedPitaya which zynq7 based device so I am doing the following:

```
$ cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake/platforms/zynq7_generic.cmake                                                                                                               

-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: zynq7
-- C_FLAGS : -mcpu=cortex-a9 -mfpu=vfpv3 -mfloat-abi=hard -Wall -Wextra
-- Found LIBMETAL: /usr/local/lib/libmetal.so  
-- Looking for include file stdatomic.h
-- Looking for include file stdatomic.h - found
-- Looking for include file fcntl.h
-- Looking for include file fcntl.h - found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/open-amp/build-baremetal
```
Then I do the `make install` but it is complaining about not finding some of the xilinx files (xscugic.h) . I don't know whether these files are generated when we import an hdf files or these are just generic files. So do we need to modify the cmake files to include some other directories or whats the method here? Below is the result of running make:

```
$ make VERBOSE=1 DESTDIR=install_dir/ install
/home/waqar/tools/Xilinx/SDK/2017.2/tps/lnx64/cmake-3.3.2/bin/cmake -H/home/user/libmetal/libmetal -B/home/user/libmetal/libmetal/build-baremetal --check-build-system
 CMakeFiles/Makefile.cmake 0
/home/waqar/tools/Xilinx/SDK/2017.2/tps/lnx64/cmake-3.3.2/bin/cmake -E cmake_progress_start /home/user/libmetal/libmetal/build-baremetal/CMakeFiles /home/waqar/workspace/MI
T/libmetal/build-baremetal/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/user/libmetal/libmetal/build-baremetal'
make -f lib/CMakeFiles/metal-static.dir/build.make lib/CMakeFiles/metal-static.dir/depend
make[2]: Entering directory '/home/user/libmetal/libmetal/build-baremetal'
cd /home/user/libmetal/libmetal/build-baremetal && /home/waqar/tools/Xilinx/SDK/2017.2/tps/lnx64/cmake-3.3.2/bin/cmake -E cmake_depends "Unix Makefiles" /home/waqar/workspa
ce/MIT/libmetal /home/user/libmetal/libmetal/lib /home/user/libmetal/libmetal/build-baremetal /home/user/libmetal/libmetal/build-baremetal/lib /home/waqar/works
pace/MIT/libmetal/build-baremetal/lib/CMakeFiles/metal-static.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/user/libmetal/libmetal/build-baremetal'
make -f lib/CMakeFiles/metal-static.dir/build.make lib/CMakeFiles/metal-static.dir/build
make[2]: Entering directory '/home/user/libmetal/libmetal/build-baremetal'
[  5%] Building C object lib/CMakeFiles/metal-static.dir/dma.c.obj
cd /home/user/libmetal/libmetal/build-baremetal/lib && /home/waqar/tools/Xilinx/SDK/2017.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin/arm-none-eabi-gcc  -DDEFAULT_LOGGER_ON -DME
TAL_INTERNAL -mcpu=cortex-a9 -mfpu=vfpv3 -mfloat-abi=hard -g -I/home/user/libmetal/libmetal/build-baremetal/lib/include    -Wall -Werror -Wextra -o CMakeFiles/metal-static.
dir/dma.c.obj   -c /home/user/libmetal/libmetal/lib/dma.c
In file included from /home/user/libmetal/libmetal/build-baremetal/lib/include/metal/system/generic/sys.h:28:0,
                 from /home/user/libmetal/libmetal/build-baremetal/lib/include/metal/sys.h:81,
                 from /home/user/libmetal/libmetal/build-baremetal/lib/include/metal/io.h:22,
                 from /home/user/libmetal/libmetal/build-baremetal/lib/include/metal/device.h:16,
                 from /home/user/libmetal/libmetal/lib/dma.c:9:
/home/user/libmetal/libmetal/build-baremetal/lib/include/metal/system/generic/./zynq7/sys.h:17:21: fatal error: xscugic.h: No such file or directory
 #include "xscugic.h"
                     ^
compilation terminated.
lib/CMakeFiles/metal-static.dir/build.make:62: recipe for target 'lib/CMakeFiles/metal-static.dir/dma.c.obj' failed
make[2]: *** [lib/CMakeFiles/metal-static.dir/dma.c.obj] Error 1
make[2]: Leaving directory '/home/user/libmetal/libmetal/build-baremetal'
CMakeFiles/Makefile2:93: recipe for target 'lib/CMakeFiles/metal-static.dir/all' failed
make[1]: *** [lib/CMakeFiles/metal-static.dir/all] Error 2
make[1]: Leaving directory '/home/user/libmetal/libmetal/build-baremetal'
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

Cheers.
Waqar

edmooring, 2019-11-18:
The missing files are part of the Xilinx BSP. You would need to include the path to your BSP include directories in your CMAKE_C_FLAGS settings. I suspect you will also need to add a link directory pointing to the BSP lib directory.

waqarrashid33, 2019-11-24:
Hello Mr @edmooring,
Thank you for your answer.
I am no longer actively working on this project anymore. I still have some documentation to submit so I will make sure to include your answer there. Will try it on my own but I would have to do the whole Xilinx setup again and I probably won't have much time to do it sooner.

Peace.
Waqar

closed 2021-04-28 by arnopo