OpenAMP/libmetal #101: FreeRTOS mutex does not handle priority inversion

supergaute, 2019-10-10:
Commit 10a0d5be removes the priority inheritance mechanism FreeRTOS is using with its mutexes.
Suggest to revert this commit.

edmooring, 2019-11-18:
The reasoning behind the original change is that FreeRTOS mutexes can't be statically initialized, which conflicts with the bare metal and Linux implementations. This will take some additional code restructuring.

supergaute, 2020-03-06:
Do you have any plans to do this restructuring in the near future?

arnopo, 2020-04-10:
@ed, @bentheredonethat,
I'm agree  with @supergaute. we should take benefit of FreeRTOS mutex. 

I'm trying to understand why the mutex can not be statically initialized using FreeRTOS. I'm not familiar with freertos, so don't hesitate if i missed something.
In freertos there is the following function: xSemaphoreCreateBinaryStatic [1]

This function seems to allow the possibility to have a static semaphore allocation
Therefore we could not define the METAL_MUTEX_INIT. As this macro is not used by the open-amp library, this should not affect it.

My proposal would be to  revert the 10a0d5b, and having  a static 

```
typedef struct {	typedef struct {
        StaticSemaphore_t semaphore;
	SemaphoreHandle_t m;	
} metal_mutex_t;

#define METAL_MUTEX_DEFINE(m) metal_mutex_t m;

static inline void __metal_mutex_init(metal_mutex_t *mutex)	static inline void __metal_mutex_init(metal_mutex_t *mutex)
{	{
	metal_assert(mutex);		atomic_store(&mutex->v, 0);
	mutex->m = xSemaphoreCreateBinaryStatic(mutex->semaphore);	
	metal_assert(mutex->m != NULL);	
}	}

```
I'm not able to test it,  so could someone validate if the implementation could work using FreeRTOS? 
 
[1] https://www.freertos.org/xSemaphoreCreateBinaryStatic.html

edmooring, 2020-04-15:
I'm inclined to revert the commit as well, and, since I can't find any evidence anybody is using METAL_MUTEX_INIT, deprecate it for this release, possibly with a compile time warning.

bentheredonethat, 2021-02-24:
@edmooring  @arnopo any update on this? if not,  I can send PR to revert the commit

arnopo, 2021-02-25:
I think you can send a PR to revert it, the current implementation seems not suitable for the freeRTOS. 
Just be sure that the METAL_MUTEX_INIT still exists. It is a macro, it will not possible to use the deprecated compilation attribute.

arnopo, 2021-04-06:
Hi @bentheredonethat 
Do you plan to send a PR for the next release ?

bentheredonethat, 2021-04-15:
Hi @arnopo  sorry won't have bandwidth on this one

madscientist42, 2023-03-15:
Uhm...why is this still open?

arnopo, 2023-03-15:
> Uhm...why is this still open?

You are right no reason to keep it open. If needed a new one can be opened

closed 2023-03-15 by arnopo

madscientist42, 2023-03-15:
Heh.  Thank you.  If LOOKED like it might've been fixed but wasn't clear
and I'm about to start doing integration work and needed to know to tell a
client there might be gotchas.