OpenAMP/libmetal #106: 2020.01: test suite is failing

kloczek, 2020-03-07:
```
[tkloczko@barrel libmetal-2020.01]$ make test
Running tests...
/usr/bin/ctest --force-new-ctest-process
Test project /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01
    Start 1: test-metal-shared
1/1 Test #1: test-metal-shared ................***Failed    0.43 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.44 sec

The following tests FAILED:
          1 - test-metal-shared (Failed)
Errors while running CTest
make: *** [Makefile:133: test] Error 8
```

arnopo, 2020-04-06:
Hi  @kloczek,
My apologize, i missed your issue report...

Do you solve your issue?
i ran them on my Linux PC with success, using following commands:
cmake . -WITH_TESTS -With_DOC -WITH_EXAMPLES -Bbuild
cd build/
make
make test

If you still have the issue please could you share share the logs available here:
build/Testing/Temporary

Thanks,
Arnaud

kloczek, 2020-04-06:
Just added those option and now I found that something is wrong with install target:
```
+ /usr/bin/make install DESTDIR=/home/tkloczko/rpmbuild/BUILDROOT/libmetal-2020.01-2.fc33.x86_64 'INSTALL=/usr/bin/install -p'
/usr/bin/cmake -S/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 -B/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/CMakeFiles /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/CMakeFiles/progress.marks
/usr/bin/make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
/usr/bin/make -f lib/CMakeFiles/metal-shared.dir/build.make lib/CMakeFiles/metal-shared.dir/depend
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
cd /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/lib /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/lib /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/lib/CMakeFiles/metal-shared.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
/usr/bin/make -f lib/CMakeFiles/metal-shared.dir/build.make lib/CMakeFiles/metal-shared.dir/build
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
make[2]: Nothing to be done for 'lib/CMakeFiles/metal-shared.dir/build'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
[ 31%] Built target metal-shared
/usr/bin/make -f test/CMakeFiles/metal-headers.dir/build.make test/CMakeFiles/metal-headers.dir/depend
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
cd /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test/CMakeFiles/metal-headers.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
/usr/bin/make -f test/CMakeFiles/metal-headers.dir/build.make test/CMakeFiles/metal-headers.dir/build
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
make[2]: Nothing to be done for 'test/CMakeFiles/metal-headers.dir/build'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
[ 77%] Built target metal-headers
/usr/bin/make -f test/CMakeFiles/test-metal-shared.dir/build.make test/CMakeFiles/test-metal-shared.dir/depend
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
cd /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test/CMakeFiles/test-metal-shared.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
/usr/bin/make -f test/CMakeFiles/test-metal-shared.dir/build.make test/CMakeFiles/test-metal-shared.dir/build
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
make[2]: Nothing to be done for 'test/CMakeFiles/test-metal-shared.dir/build'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
[100%] Built target test-metal-shared
/usr/bin/make -f doc/CMakeFiles/doc.dir/build.make doc/CMakeFiles/doc.dir/depend
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
cd /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/doc /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01 /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/doc /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/doc/CMakeFiles/doc.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
/usr/bin/make -f doc/CMakeFiles/doc.dir/build.make doc/CMakeFiles/doc.dir/build
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
cd /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/doc && /usr/bin/doxygen /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/doc/Doxyfile
warning: Tag 'CLANG_ASSISTED_PARSING' at line 996 of file '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/doc/Doxyfile' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_OPTIONS' at line 1004 of file '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/doc/Doxyfile' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'PERL_PATH' at line 2075 of file '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/doc/Doxyfile' has become obsolete.
[... cut ...]
make[1]: Entering directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
make[1]: Nothing to be done for 'preinstall'.
make[1]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01'
Install the project...
/usr/bin/cmake -P cmake_install.cmake
-- Install configuration: "Debug"
CMake Error at lib/cmake_install.cmake:41 (file):
  file INSTALL cannot find "/usr/include/metal/alloc.h": No such file or
  directory.
Call Stack (most recent call first):
  cmake_install.cmake:42 (include)
```

kloczek, 2020-04-06:
ctest still is failing:
```
[tkloczko@barrel libmetal-2020.01]$ ctest -VV
UpdateCTestConfiguration  from :/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/DartConfiguration.tcl
Test project /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: test-metal-shared

1: Test command: /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test/test-metal-shared
1: Test timeout computed to be: 10000000
1: metal: debug:     added page size 4096 @/tmp
1/1 Test #1: test-metal-shared ................***Failed    0.36 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.36 sec

The following tests FAILED:
          1 - test-metal-shared (Failed)
Errors while running CTest
```

kloczek, 2020-04-06:
install is broken by my not finished patch which moves from using your custom install paths to thwose provided by `GNUInstallDirs` module.

Nevertheless test suite seems still is broken.

arnopo, 2020-04-07:
Your issue seems linked to you environment. I can not reproduce it on my Ubuntu 18.04.

it seems for instance that the CMAKE_INSTALL_PREFIX is set

could you:
- give me more information about your system:
   - target  processor family
   - compiler
   - PC windows or Linux with OS version
   - cmake version

- have a look in  logs available in Testing/Temporary/LastTest.log  to have more detail on the issue...

- try to perform the test using default settings (CMAKE_INSTALL_PREFIX an other not customized)?

kloczek, 2020-04-07:
-  x86_64
- gcc 10.0.1 from fedora rawhoide
- Linux kernel 5.6.x
- cmake 3.17.0

Content of the Testing/Temporary/LastTest.log
```
Start testing: Apr 07 09:31 BST
----------------------------------------------------------
1/1 Testing: test-metal-shared
1/1 Test: test-metal-shared
Command: "/home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test/test-metal-shared"
Directory: /home/tkloczko/rpmbuild/BUILD/libmetal-2020.01/test
"test-metal-shared" start time: Apr 07 09:31 BST
Output:
----------------------------------------------------------
metal: debug:     added page size 4096 @/tmp
<end of output>
Test time =   0.38 sec
----------------------------------------------------------
Test Failed.
"test-metal-shared" end time: Apr 07 09:31 BST
"test-metal-shared" time elapsed: 00:00:00
----------------------------------------------------------

End testing: Apr 07 09:31 BST
```
Lets move on the side for now all bits related to install process and focus only on test suite :)

arnopo, 2020-04-07:
> * x86_64
> * gcc 10.0.1 from fedora rawhoide
> * Linux kernel 5.6.x
> * cmake 3.17.0
>
> Content of the Testing/Temporary/LastTest.log
>
> ```
> Start testing: Apr 07 09:31 BST [...]

I would say no issue if i focus only on the test suite :)

$ make test
Running tests...
Test project /local/home/arnopo/views/openamp/libmetal/build
    Start 1: test-metal-shared
1/2 Test #1: test-metal-shared ................   Passed    0.58 sec
    Start 2: test-metal-static
2/2 Test #2: test-metal-static ................   Passed    0.64 sec

100% tests passed, 0 tests failed out of 2

Total Test time (real) =   1.23 sec

So that's why i focus on the build environment...

cmake 3.17 is a good candidate ( i'm using cmake 3.13)  it introduced some regressions in zephyr (https://github.com/OpenAMP/libmetal/issues/107) 
i will perform test using the cmake 3.17

arnopo, 2020-04-07:
Same result with cmake 3.17 test passed on my side.

Concerning your logs, it seems that something fails in metal_init .
Here is my logs:
```
1: Test command: /local/home/arnopo/views/openamp/libmetal/build/test/test-metal-shared
1: Test timeout computed to be: 10000000
1: metal: debug:     added page size 4096 @/tmp
1: metal: warning:   skipped page size 2097152 - invalid args
1: metal: warning:   skipped page size 1073741824 - invalid args
1: metal: debug:     registered platform bus
[...]
```
look like metal_linux_bus_init returns an error... 
As there is not explicit message on error, difficult to identify the issue... 

So could you check if metal_linux_bus_init returns an error, and then isolate the issue in code by adding some traces on error in functions or by using a debug tool?

kloczek, 2020-04-07:
I'm using in cmake options `-D WITH_STATIC_LIB=OFF` (I don't need static library).

arnopo, 2020-04-09:
Any time to do more test on your side? adding trace or debug, as requested?

FYI i have implemented a github action  to execute these tests on PR. Please have a look here.
https://github.com/OpenAMP/libmetal/pull/111/checks?check_run_id=571253239
Tests are passed.
So we have to figure out why your environment fails these tests.

kloczek, 2020-04-09:
Will try to spend on that some time tomorrow :)

FYI I'm building everything with LTO and -Wl,--as-needed.

kloczek, 2020-05-10:
Forgot to update (sorry).

Looks like 2020.04.0 is ok now.

Closing

closed 2020-05-10 by kloczek