OpenAMP/libmetal #117: Time unit mismatch in FreeRTOS __metal_sleep_usec

nsullivan33, 2020-05-04:
The math in sleep.h for FreeRTOS seems suspect, I think it might create delays 1,000 times larger than the caller expects:

`static inline int __metal_sleep_usec(unsigned int usec)
{
	const TickType_t xDelay = usec / portTICK_PERIOD_MS;

	vTaskDelay(xDelay);
	return 0;
}`

Shouldn't it be "(usec / 1000) / portTICK_PERIOD_MS" to convert microseconds to ticks?

arnopo, 2020-06-23:
Sorry for the late answer...
you are probably right regarding the APIs. Are you able to verify it in your environment?
Else you can propose the patch, we will find a way to verify it...

nsullivan33, 2020-06-25:
I can verify it and create a patch.  Do you need me to do both?  I think just dividing the "usec" variable by 1000, to convert to milliseconds, should be enough though the resolution would be poor.

arnopo, 2020-06-29:
If you can do both it is perfect. 
Concerning the calculation.  the precision will be one tick so dividing by 1000 is ok. 
And i think that we also add an extra  tick to the final result to ensure that we at least wait for the expected time..
"xDelay = ((usec / 1000) / portTICK_PERIOD_MS) + 1"
thanks!

arnopo, 2020-08-28:
@nsullivan33 
Do you plan to send a patch has you proposed?

nsullivan33, 2020-08-28:
Sorry, I've been incredibly busy lately.  I likely won't have time for a while...

Thanks,
Nathan Sullivan

edmooring, 2020-10-02:
Taking a look at it:
The following small patch:

diff --git a/test/system/freertos/sleep.c b/test/system/freertos/sleep.c
index 9bc72d1..6964cd3 100644
--- a/test/system/freertos/sleep.c
+++ b/test/system/freertos/sleep.c
@@ -18,13 +18,14 @@
 static int sleep(void)
 {
        int rc;
-       unsigned int usec = 3;
+       unsigned int usec = 300000;
        unsigned long long tstart, tend, tdelayed;
 
        tstart = metal_get_timestamp();
        metal_sleep_usec((usec/portTICK_PERIOD_MS));
        tend = metal_get_timestamp();
        tdelayed = tend - tstart;
+       metal_log(METAL_LOG_ERROR, "tdelayed = %llu, usec/portTICK_PERIOD_MS = %lu\r\n", tdelayed, usec/portTICK_PERIOD_MS);
        if (tdelayed > (usec/portTICK_PERIOD_MS))
                rc = -1;
        else

gives the following output:
metal: error:     tdelayed = 3000, usec/portTICK_PERIOD_MS = 30000
metal: info:      result [sleep]............................. pass

This is clearly not right.

metal_get_timestamp() on FreeRTOS returns the number of TICKS since the scheduler started. This doesn't match the Linux implementation, which returns the number of nano-seconds since the epoch. The test code is expecting the result to be in microseconds. The documentation doesn't specify the units.

The test code is wrong, because it does usec/ticks, duplicating the action performed in __metal_sleep_usec() for FreeRTOS, and because a value of tdelayed that is off by an order of magnitude is not a pass.

arnopo, 2020-10-02:
Right,  There is nothing  generic is metal_get_timestamp...

Seems that this API is only use for test. But could also be used for traces or by OpenAMP clients.

If we want to have a generic interface the out should be in µs to be aligned with the sleep but also to be compatible with baremetal. 

What about  creating a new API  "unsigned long long  metal_get_timestamp_usec(void)" to clarify this?
We can migrate the FreeRTOS, NUTTX, and Linux tests on it.
Some reworks will be needed for Zephyr ( we can raise an issue for this)
In a second, we could  step put metal_get_timestamp in deprecated API.

edmooring, 2020-10-03:
Just to fix this issue, without changing the API, I created https://github.com/OpenAMP/libmetal/pull/129.
What do you think? I have tested this, and it works for FreeRTOS.

I like your proposed fix, but it might take too long to fit it into the upcoming release. We would need testers for Zephyr and Nuttx.

arnopo, 2020-10-06:
> Just to fix this issue, without changing the API, I created #129.
> What do you think? I have tested this, and it works for FreeRTOS.
> 
> I like your proposed fix, but it might take too long to fit it into the upcoming release. We would need testers for Zephyr and Nuttx.

Ok that seems reasonable, let's do this in several steps

closed 2023-10-24 by github-actions[bot]