OpenAMP/libmetal #146: libmetal on Linux without kernel modules (uio_pdrv_genirq etc.)?

infn-ke, 2021-03-09:
Is it supported to run libmetal on Linux without having the kernel modules uio_pdrv_genirq, uio_dmem_genirq, vfio-pci and uio_pci_generic? The code tries to do `modprobe` on these in `metal_linux_probe_driver`. I want to use openamp on Linux but without being tied to those kernel modules.

edmooring, 2021-03-09:
Unfortunately, there is no way to do that right now. De-coupling the libmetal device abstraction on linux from UIO would be challenging.

infn-ke, 2021-03-10:
I tried to decouple it by replacing the `metal_sys_init` with an implementation that simply registers the generic bus `metal_generic_bus` rather than doing modprobe on all the kernel modules. Then I just register a generic device describing a shared memory (`metal_register_generic_device`). Wouldn't that be a viable path?

edmooring, 2021-03-31:
Sorry about the delay in responding. That might work. I may need something similar, so I will probably try something like it.

closed 2021-04-28 by arnopo