OpenAMP/libmetal #156: Let the compiler do inlining

ollie-etl, 2021-05-17:
In several places in the code [here](https://github.com/OpenAMP/libmetal/blob/04cf5d9e0ef562d676cd3f58530fed4e1901f428/lib/io.h#L297), `#defines` are used to declare functions. Is there a rational for this?

The drawbacks of this style are it:
  - removes any semblance of compiler checked type safety, 
  - makes generating FFI bindings to this library unnecessarily manual (and more error prone). This is certainly true for rust, and probably for python, although I haven't tried that recently.
  - reduces IDE tooling support.
  
I can't think of a rational for using this style - other than maybe you don't trust a compiler to inline - although that seems a pretty low bar.

I'm willing to submit PR(s) to address this, if desired

A related issue would be using consts in place of `#defines` for constants. The rational is similar, you gain type safety, tooling support, and simplify FFI generation.

edmooring, 2021-05-18:
These are all good points. I don't know of any reasonable argument against them. PRs will be welcomed.

arnopo, 2021-05-18:
Yes your proposal is certainly valid and a PR  would help to estimate the impacts.
For now, my only concern is the footprint of libmetal due to the replacement of some `#define` by constant variables...
So please separate the functions and constants updates into two separate patches.

ollie-etl, 2021-05-18:
I'm happy to separate those concerns. I've created #158 to record that