OpenAMP/libmetal #166: Support other build systems instead of cmake

pedwo, 2021-07-27:
For a lot of embedded targets, libmetal would be included into a much larger project that might not use cmake.
It would be good if the code didn't use configure_file symbols in the code like [PROJECT_SYSTEM](https://github.com/OpenAMP/libmetal/blob/master/lib/cache.h#L15).
One option could be to change the include path passed to the compiler instead of this.

arnopo, 2021-08-10:
You are right the use of config sylbole for file include in code is not enough generic. we face the issue for some IDE that do not support cmake. In this case the code has to be preconfigured to generate the include files.
If you have an alternative to propose don't hesitate to send a pull request!

nathalie-ckc, 2024-02-21:
Discussed in OpenAMP System Reference working group on 2/21/2024
We don't have other requests for solutions other than cmake and don't have plan to do this.  
This issue can be reopened if there is more motivation behind it or someone wants to contribute and maintain.