OpenAMP/libmetal #172: Shared memory use with ASAN

anon00000000, 2021-10-05:
Hello, and sorry in advance for the vague description. We use libmetal (metal_io_virt) to access shared memory on an Aarch64 Xilinx board, and when compiled with ASAN, we sometimes get strange alternating patterns of 8 bytes 0x0 then 8 bytes 0xff instead of the buffer data we usually get.

Is there anything ringing a bell about libmetal and ASAN interacting strangely?

Regards

arnopo, 2021-10-12:
Hello @anon00000000 
I never used ASAN from now 
@edmooring or @bentheredonethat, any input ?

anon00000000, 2022-04-20:
Nevermind, this was our fault; using ASAN initialized variables otherwise left uninitialized and did some funky things with our code.

closed 2022-04-20 by anon00000000