OpenAMP/libmetal #201: ERANGE usage

tammyleino, 2022-03-22:
Is there a reason a new error code, METAL_ERANGE, was not created instead of using -ERANGE?

tammyleino, 2022-03-22:
I see that the METAL_XYZ return codes are used for routines that return unsigned values and EINVAL and ERANGE are used for routines that return int.  Has the possible case been ruled out for the unsigned routines that the error code and valid data would overlap and the user would think the routine was not successful?

arnopo, 2022-03-30:
Please could you provide some concrete examples to illustrate your point?

arnopo, 2025-02-03:
@tammyleino: I tried to find the code you mentioned without success. As it is an old issue, perhaps it is no longer valid. Could you point me to the issue? Otherwise, I will close the issue.

closed 2025-02-03 by arnopo