OpenAMP/libmetal #257: Restructure content in README.md

wmamills, 2023-10-10:
The README is a bit log and rambling.

The README is also the main page for the doygen docs which make it one big page also.

Issues/Actions
* Library categogy data is good but should be in the intro to each section
* * move this content into per module files in doc/ and replace by a table in README.
* Contributing info does not need to be on the main page, make a page and link it from README
* Simply build instructions on the main page (README) and move the detailed instructions to a sub-page.
* * Example1: Most people that build for Zephyr will use the Zephyr integration and it will just work.  Move the details for the other 5% off the main page.
* * Example2: Build for bare-metal, use an existung generic target on main page.  Move instucions on make a new custom target to new page and suggest they carrry that def in their build system instead of pushing to libmetal.
* (Example2b: ceva CPU port is just generic.  Make it easy to support a new CPU that is generic.)

arnopo, 2026-10-23:
Hi @sipke,
if you have time to address it...

sipke, 2026-10-26:
Was able to address most of this with https://github.com/OpenAMP/libmetal/pull/376

closed 2026-11-10 by arnopo