OpenAMP/libmetal #268: shmem-provider header file

gabrielrodcanal, 2023-10-15:
Hi there,
I am trying to build the driver for AMD Xilinx AIE and it needs the shmem-provider.h file from libmetal. Currently that file doesn't exist and there is only shmem.h. I can imagine the file was renamed at some point or its funcionality might have merged into another files... which version of libmetal has this file?

Thank you.

tnmysh, 2023-10-16:
I am not sure. I checked version v2018.04 and it still has shmem.h.
It's not recommended to get to that release.

Some more details can be helpful. Such as are you using petalinux ? what version?

tnmysh, 2023-10-16:
Also I believe it was always shmem.h. Here is initial commit : https://github.com/OpenAMP/libmetal/commit/7bda1db4a90d230f5c29fe8aa1e4454238122d05

arnopo, 2026-10-23:
@gabrielrodcanal,
I’m closing this issue because it has been stale for a long time. If you are still facing the issue, please reopen it with the details requested by @tnmysh.

closed 2026-10-23 by arnopo