OpenAMP/libmetal #295: CI: Pin to specific version of Zephyr SDK in default PRs

wmamills, 2024-05-05:
Unlike open-amp CI, this script uses the very latest version of the Zephyr SDK on each PR check.
This is bad.  While it is good to test against the latest SDK, that should not be mixed into checking a PR.

The default PR should use a pinned version of the SDK (and the same one as open-amp).

It would be good to have separate CI jobs that would test a known good library (current main should be fine) against the latest Zephyr SDK, latest Zephyr, latest Ubuntu etc

wmamills, 2024-05-06:
@arnopo what do you want to do about this?  Right now libmetal is using 0.16.6 for CI and can change at any time.  open-amp is using 0.16.1 (which is old but at least a fixed version.)

wmamills, 2024-08-14:
This issue caused problems again in PR #302 .
If CI randomly fails for reasons unrelated to the CI, it just teaches people to ignore the CI results.
Alternatively if Upstream Zephyr has a breaking change and we have no PRs, we don't know about it until the next PR.

Testing libmetal against latest Zephyr upstream should be its own job and should be run weekly using libmetal main.
Testing in PRs and merges should be against a known good version of Zephyr and Zephyr SDK.

arnopo, 2024-08-26:
> This issue caused problems again in PR #302 . If CI randomly fails for reasons unrelated to the CI, it just teaches people to ignore the CI results. Alternatively if Upstream Zephyr has a breaking [...]

Seems reasonable. What I cannot understand is why it fails with Zephyr 3.7.99 as this is the last release of zephyr.

PR #307 (2024-09-20): CI: Do Zephyr build tests on known good and latest versions
Allow Zephyr testing to either use locked known good values or the very latest versions.

The known good versions are best for PR checking as if the build fails it is almost always the PR itself that broke it.

The latest version is good for periodically checking compatibility with the very latest Zephyr changes.

For now we run both on pushes and PRs.

We also run main against the latest zephyr check weekly as a look ahead.

Fixes: #295
closed 2024-09-20 by arnopo