OpenAMP/libmetal #323: libmetal: Buffer overflow in metal_sys_io_mem_map()

iuliana-prodan, 2025-02-05:
In `void metal_sys_io_mem_map(struct metal_io_region *io)` ([1] and [2]) if the I/O region `size` is a multiple of `1<<page_shift` will result in a buffer overflow here, in the `for` loop:
```
for (p = 0; p <= (io->size >> io->page_shift); p++) {
	metal_machine_io_mem_map(va, io->physmap[p],
				 psize, io->mem_flags);
	va += psize;
}
```
 
The _solution_ is:
```
diff --git a/libmetal/lib/system/freertos/io.c b/libmetal/lib/system/freertos/io.c
index 319322c..dde8242 100644
--- a/libmetal/lib/system/freertos/io.c
+++ b/libmetal/lib/system/freertos/io.c
@@ -22,7 +22,7 @@ void metal_sys_io_mem_map(struct metal_io_region *io)
        if (psize) {
                if (psize >> io->page_shift)
                        psize = (size_t)1 << io->page_shift;
-               for (p = 0; p <= (io->size >> io->page_shift); p++) {
+               for (p = 0; p <= ((io->size -1) >> io->page_shift); p++) {
                        metal_machine_io_mem_map(va, io->physmap[p],
                                                 psize, io->mem_flags);
                        va += psize;
diff --git a/libmetal/lib/system/generic/io.c b/libmetal/lib/system/generic/io.c
index 966bfc5..ac36f09 100644
--- a/libmetal/lib/system/generic/io.c
+++ b/libmetal/lib/system/generic/io.c
@@ -22,7 +22,7 @@ void metal_sys_io_mem_map(struct metal_io_region *io)
        if (psize) {
                if (psize >> io->page_shift)
                        psize = (size_t)1 << io->page_shift;
-               for (p = 0; p <= (io->size >> io->page_shift); p++) {
+               for (p = 0; p <= ((io->size -1) >> io->page_shift); p++) {
                        metal_machine_io_mem_map(va, io->physmap[p],
                                                 psize, io->mem_flags);
                        va += psize;
```
 
[1] https://github.com/OpenAMP/libmetal/blob/main/lib/system/freertos/io.c#L14
[2] https://github.com/OpenAMP/libmetal/blob/main/lib/system/generic/io.c#L14

arnopo, 2025-02-05:
Seems to me correct,  don't hesitate to directly send a PR to propose the solution

iuliana-prodan, 2025-02-11:
> Seems to me correct, don't hesitate to directly send a PR to propose the solution

Fix sent for review: https://github.com/OpenAMP/libmetal/pull/324

closed 2025-02-14 by arnopo