OpenAMP/libmetal #326: CI: "Zephyr build from lastest on mail " fails

arnopo, 2025-02-13:
Since the merge of https://github.com/zephyrproject-rtos/zephyr/commit/4c1275d63aa31faf293a70fc4332b84eb72f5e4f the CI is failed.

```
CMake Error at zephyrproject/zephyr/cmake/modules/version.cmake:66 (string):
  Syntax error in cmake code at

    /github/workspace/zephyrproject/zephyr/cmake/modules/version.cmake:66

  when parsing string

    EXTRAVERSION = ([a-z0-9\.\-]*)

  Invalid escape sequence \.
Call Stack (most recent call first):
  zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfigVersion.cmake:59 (include)
  cmake/syscheck.cmake:8 (find_package)
  CMakeLists.txt:17 (include)

```

The issue seems linked to the syntax in the cmake/modules/version.cmake file

A fix has been proposed in zephyr: 
https://github.com/zephyrproject-rtos/zephyr/pull/85724

arnopo, 2025-02-13:
Analysis from @tejlmand in https://github.com/zephyrproject-rtos/zephyr/pull/85724#pullrequestreview-2614623635
shows that the issue in In OpenAMP that not well manage cmake policy

I can see two options:

1- Increase the minimum version of CMake in [libmetal's CMakeLists.txt](https://github.com/OpenAMP/libmetal/blob/main/CMakeLists.txt).

- The current minimum CMake version is 3.0.2 (Dec 3, 2018).
- Zephyr requests a minimum CMake version of 3.20 (Apr 8, 2021).
- This could be an opportunity to remove the management of old policies CMP0048 and CMP0077.

2- Activate the CMP0053 policy in [libmetal's CMakeLists.txt](https://github.com/OpenAMP/libmetal/blob/main/CMakeLists.txt).
- This policy was introduced in CMake 3.1.
```
Cmake
if (POLICY CMP0053)
  cmake_policy(SET CMP0053 NEW)
endif()
```
I would be in favor of migrating to CMake 3.20.

@tnmysh  @edmooring  @wmamills and other do you see an issue for migrating to cmake 3.20?

wmamills, 2025-02-13:
@arnopo Ubuntu 20.04 which is still supported has cmake 3.16.3
of course you can install a newer one with pip or from the kitware archive and you would need to for zephyr.
However if not using Zephyr it would be nice to be compatible with the one from 20.04

Can we use 3.16 as a minimum?

(debian bullseye aka oldstable has 3.18.4)

arnopo, 2025-02-14:
> Actually Ubuntu 20.04 is public EOL in April of this year. It is still supported for pro customers until April 2030. Ubuntu 22.04 has cmake version 3.22.1 so 3.20 would be OK.
> 
> I am ok with either 3.16 or 3.20 as min version

Version 3.16 is OK. Regarding CMake policies, we are compatible with CMake 3.13 and higher.

arnopo, 2025-02-14:
I pushed a PR #327  that update to 3.16 ,  I also tested for open-amp and I will push it later if everyone is OK to go in this direction

wmamills, 2025-02-15:
@arnopo why did my open-amp PR#636 pass all tests including the latest zephyr test??
https://github.com/OpenAMP/open-amp/pull/636

arnopo, 2025-02-17:
> [@arnopo](https://github.com/arnopo) why did my open-amp PR#636 pass all tests including the latest zephyr test?? [OpenAMP/open-amp#636](https://github.com/OpenAMP/open-amp/pull/636)

In libmetal, we use the `cmake` command to build libmetal for different architectures. In OpenAMP CI, we are using the `west` command. The Zephyr build chain seems to handle the check of the VERSION file present in libmetal and open-amp differently depending on the command used.

closed 2025-02-24 by arnopo